josefprusa/Prusa3 #94: x-ends are mirrored

foulowl, 2015-08-05:
From complete-printer.scad: (correct)
http://i.imgur.com/uO5GQ6A.png

From output folder: (incorrect)
http://i.imgur.com/SSK80aW.png

Solution: mirror x-end across x axis.

Joaz, 2015-08-08:
They actually print like this in the output folder. You turn them around. The open side of the bearings faces the frame. I think the complete-printer.scad is outdated.

foulowl, 2015-08-08:
Ah, well it needs to be fixed either way, since it's internally inconsistent. 

Is there a picture of the correct orientation?

Thank you!

Joaz, 2015-08-09:
http://ext01.reprapsource.com/reprapsource-Prusa-i3-Kit-Build-Manual/Prev/IMG_0627.JPG

AxTheB, 2015-08-10:
Readme in box_frame says "Using doc/complete-printer-plus.scad as a visual clue and doc/manual.txt as a guide, build the printer." and in the orientation in doc/complete-printer-plus.scad is correct iirc.

closed 2015-08-10 by AxTheB