josefprusa/Prusa3 #96: WARNING: Ignoring unknown module 'x_beltclamp'

quillog, 2016-01-20:
Cant load the file correctly:

File: Pursa3\box_frame\doc\complete-printer-plus.scad

Software: OpenSCAD version 2015.03-2 

LOG:

Module cache size: 9 modules
Compiling design (CSG Tree generation)...
ECHO: 320
ECHO: 320
WARNING: Ignoring unknown module 'x_beltclamp'. 
WARNING: Ignoring unknown module 'x_beltclamp'. 
Compiling design (CSG Products generation)...
Geometries in cache: 770
Geometry cache size in bytes: 1940008
CGAL Polyhedrons in cache: 1271
CGAL cache size in bytes: 104843040
Compiling design (CSG Products normalization)...
Compiling background (37 CSG Trees)...
Normalized CSG tree has 1459 elements
Compile and preview finished.
Total rendering time: 0 hours, 1 minutes, 1 seconds

jakevossen5, 2016-12-11:
Same issue. Have you found a solution?

SinoSnow, 2016-12-12:
I threw this fight. And sewed the old version of the firmware (marlin 1.0.2-1) from the old version 023 of the arduino. And all went.

MisterTobi, 2017-02-15:
well done!